GEOUNED-org/GEOUNED #2: Request for license file

shimwell, 2023-10-06:
Dear uned, really keen to make us of this package 

Would it be possible to add a license file to the repo root as this impacts how and if we can use this at work.

Many thanks for sharing the code, this has huge potential

jpcatalanUNED, 2023-10-06:
Dear Jonathan,

I have just include the license file. Thank you for your comments.

Regards

closed 2023-10-06 by jpcatalanUNED

makeclean, 2023-10-10:
Hi @jpcatalanUNED any chance you could make the licence slightly more permissive? Currently a GPL-3.0 licence (typically) prohibits commercial use, not an issue for me, but others such as the private fusion companies would likely make use of this tool if you switched to say a GPL-2.1 or a maybe an EUPL licence.

shimwell, 2023-10-10:
Yep for private companies it may be difficult to make use of a code with GPL3.0 license. As I understand it users would risk needing to open up the rest of our software stack than connects to the code.

[Bullet point number 5 in this summary of GPL-3.0 ](https://gist.github.com/kn9ts/cbe95340d29fc1aaeaa5dd5c059d2e60 )
"5. If you dare build your business solely from this code, you risk open-sourcing the whole code base."

Here is a nice write up on how software licenses combine, licenses on the left are more permissive and I can see that the licenses @makeclean mentions are more to the left than GPL3.0
https://dwheeler.com/essays/floss-license-slide.html 
[image]