GEOUNED-org/GEOUNED #11: Labels and materials are not read correctly for some STEPS

MigMagan, 2023-10-27:
When using STP files that are generated in Coreform Cubit, the labels are not read. This seems to be caused because these volumes do not have any elements in their InList property, as there is no higher order element that depend on them. Why some software creates STP files this way while other does not is beyond my knowledge, but they certainly seem to be valid files. 

I have solved it with some modification to LoadSTEP.LoadCAD(), and the examples still seem to run fine. However, the fact that I uncommented some lines tell me that I have probably broken something else. I am attaching a patch so that you can check it.

MigMagan, 2023-10-27:
[patch_issue11.txt](https://github.com/GEOUNED-code/GEOUNED/files/13188688/patch_issue11.txt)

psauvan, 2023-10-27:
Hi Miguel,
thank you for your contribution. I will have a look at it.
May you send me a stp example file produced by CoreForm Cubit, to see how labels look like
.

MigMagan, 2023-10-30:
Hi Patrick:
I have imported the case 3 step fie into Cubit and re-exported it. This way we can work with a well-known case (and know what we should get). 

https://www.dropbox.com/scl/fi/bso4pshfmft45els2pfam/case3_Cubit.stp?rlkey=32kdj8kjpztnah1tnk92kycrg&dl=0