GEOUNED-org/GEOUNED #15: Reverse Function inFormat?

mxmws, 2024-02-14:
I am trying to run the reverse function.

I am using this ini file. I am not sure why inFormat is supposed to be mcnp or openMC_XML, as I thought the purpose of this function was converting from step to csg.

```[Setting]
inputFile = GEOUNED/testing/inputSTEP/biseau.stp
CADFile = modelCAD

# CSG format allowed are mcnp or openMC_XML
inFormat = mcnp      

# box dim in cm
# box entries xmin xmax ymin ymax zmin zmax
outBox = -500 500 -500 500 0 1000  

[Levels]
UStart = 0
levelMax = all

# For cell and materials "rangeType" value may be all, include, exclude
# range example is : 2:20, 34, 38, 100:200
[Cells]
rangeType = all
#range = 1:13

[Materials]
rangeType = exclude
range = 0

[Options]
splitTolerance = 0.01
```

This is the last part of the error message I get when trying to run the code: `.../GEOUNED/src/GEOReverse/Modules/Parser/parser.py", line 295, in _protect_nums cell, mat, rho = tokens[:3] ValueError: not enough values to unpack (expected 3, got 1) `. Basically at this point the variable `input` of the class `card` is just `['HEADER;']`. The issue seems to be related to material, but I wish to ignore material.

psauvan, 2024-02-14:
The ini file you are using is for convertion from MCNP/OpenMC to step file.

If you want to convert from step to CSG (MCNP or OpenMC) you have to use the other format for the inifile 
https://github.com/GEOUNED-code/GEOUNED/blob/main/scripts/config.ini

mxmws, 2024-02-14:
Thank you for your reply.
So I need to use the `config.ini` for the reverse function instead of the `configReverse.ini`?
I just tried this and it just prints out `Files bad section name`. What can I do?