GEOUNED-org/GEOUNED #17: Materials not populating from material file

Teancum-Q, 2024-02-15:
Hi,
 My materials are labeled in Freecad with label names containing "_m01_" or the like, but these volumes are not labeled with the appropriate materials when output.  I am calling the correct materials file, and it is in the correct format. Though i've tried with and without a leading 0.  I'm just not sure if it's a bug or something is incorrect in my execution. I have attached the relevant files below. Thank you for the help. 
I am also Using the DEV branch of the code. If that matters. 

[materials.zip](https://github.com/GEOUNED-code/GEOUNED/files/14303330/materials.zip)

psauvan, 2024-02-16:
I don't what happens with your translation, it is very strange. I have translated your model on my computer (with the same sources you are using and the same FreeCAD version) and the materials in the cells are correctly written. 
I have asked a colleague to run the translation on its computer and he got also a correct result.

I also workaround to try to catch your error but I didn't succeed. 
What I can suggest is to try with another version of FreeCAD. 

Which OS do you use to  run the translation?  I have carried out the tests on Windows OS.  

[Gegi..mcnp.txt](https://github.com/GEOUNED-code/GEOUNED/files/14313706/Gegi.mcnp.txt)

Teancum-Q, 2024-02-29:
Well it's good to know it's working on your end. I'm running Freecad 0.21.2 on windows 10.

psauvan, 2024-02-29:
you may try the FreeCAD 0.20.2 version (the one I currently use)

Teancum-Q, 2024-03-01:
Hi, I will try that. I did find that if you name two bodies the same name, Freecad makes a hidden name append to the object. This is usually a duplicate of the name after a period i think. This could have caused the issue. I'll have to do some checking to make sure.

closed 2024-06-14 by alexvalentine94