GEOUNED-org/GEOUNED #26: Misshaped Translation

Teancum-Q, 2024-02-29:
Hey Guys,
  Thanks for your help before. I'm trying to get this relatively simple cad geometry (mostly cylinders and squares) to translate properly, but it seems to distort the shape pretty drastically. I assume this is a setting i have wrong, would you mind running it and see if you get the same error. It could be linked back to may last question, and i need to change my version of freecad. Thank  you for your help. 

I've attached the stp file, the geouned result, and the input deck.

[New folder.zip](https://github.com/GEOUNED-code/GEOUNED/files/14441265/New.folder.zip)

psauvan, 2024-02-29:
I don't understand what is your issue.
I have plot your output with MCNP plotter and it looks fine. I run it and checked statistically the solid volumes and they are correct (the simulation doesn't lose particle either)
Finally I translated back to CAD your model MCNP and it looks like the original.
I attach the MCNP simulation output as well as the back converted MCNP model
[Assembly.zip](https://github.com/GEOUNED-code/GEOUNED/files/14446713/Assembly.zip)

Teancum-Q, 2024-03-01:
Hi guys,
 I think i actually made a mistake on my end. My MCNP visualizer just wasn't set to high enough resolution. I was using gxs_view.

closed 2024-03-01 by Teancum-Q