GEOUNED-org/GEOUNED #27: Cylinder displayed incorrectly

Teancum-Q, 2024-03-01:
Hello again, 
  I'm starting to feel like you're spending all your time just answering my questions. This one is just a question about warnings. I've created a cad file similar to the last one i sent you, but one of the cylinders gets displayed incorrectly. It is roughly 10 times too long. I get a warning file, but it doesn't say why it failed, just that it might have (which it did). Is there something i can do to help this solve correctly. I've attached the relevant files, and a couple images for quick assessment.  I did try to adjust the position of the cylinder to make sure it wasn't going into another surface, but I ended up with the same phenomena. 

 Thank you for all the help. I hope that i can start implementing this tool much more regularly in my work. 

[image]
[image]
[New folder (2).zip](https://github.com/GEOUNED-code/GEOUNED/files/14454887/New.folder.2.zip)

psauvan, 2024-03-01:
There was an interfence in your orginal CAD file, that's why the translation failed.
Once the interference has been removed the CAD translate correctly
[image]
[fixed.zip](https://github.com/GEOUNED-code/GEOUNED/files/14458719/fixed.zip)

closed 2024-03-01 by psauvan

Teancum-Q, 2024-03-01:
Thank you very much. Hopefully I won't have any other questions for awhile.