GEOUNED-org/GEOUNED #28: Fill and plane

Teancum-Q, 2024-03-06:
I was wondering if you were going to implement in the mcnp parser a fully specified fill?

FILL = 0:2 1:2 0:1
4 4 2 $ i=0,1,2 for j=1 & k=0
0 4 0 $ i=0,1,2 for j=2 & k=0
0 3 3 $ i=0,1,2 for j=1 & k=1
4 4 0 $ i=0,1,2 for j=2 & k=1

or  a parser for the alternative (p, plane) using 9 points?

Form: j n P x1 y1 z1 x2 y2 z2 x3 y3 z3

I can figure out transforms for these to the other forms, might take a little work for the fill, but they are commonly used where i work. I just thought i'd ask.

psauvan, 2024-03-07:
We don't plan to implement the lattice option in GEOUNED in the near futur.
Concerning the 9 points plane definition, yes it is implemented in the current version of the code.

closed 2024-03-07 by psauvan

Teancum-Q, 2024-03-07:
Thank you for the info.