GEOUNED-org/GEOUNED #30: Conversion failures of torus related geometry

vwbus1967, 2024-03-07:
Dear all,

I'm trying to convert buried pipes and I'm facing to some conversion failures.
Would you check and give some advices?

The failures seem to relate to a torus conversion. 
To properly include the torus part, some boundary surfaces might be missing.
In addition, some torus surface signs seem not to set rightly. 

The pasted followings: 
1. pipe_swp can convert properly. (only a pipe) 
2. BaseSample fails. (several holes for burying pipes)
(3. placa3 fails, this is original from /testing/inputSTEP/Torus as the other example. )

Best Regards,

[samples.zip](https://github.com/GEOUNED-code/GEOUNED/files/14524377/samples.zip)

mxmws, 2024-03-07:
Yes, I noticed this issue too. There are multiple problems in the code. I'll describe what I found, maybe someone can help fixing it. If you use a different export than the py-file it may work. Have you tried that?

**OpenMCSurface issue for writing torus to py-file**

There is an issue with this line:
```python
coeffs ='x0={},y0={},z0={},r{},r1={},r2={}'.format(surf.Center.x, surf.Center.y, surf.Center.z, surf.MajorRadius, surf.MinorRadius, surf.MinorRadius)
```
There is a typo for the first radius and `surf.MinorRadius` is used twice. Considering the [OpenMC documentation](https://docs.openmc.org/en/latest/pythonapi/generated/openmc.XTorus.html) I think the correct line should look like this:
```python
coeffs ='x0={},y0={},z0={},a={},b={},c={}'.format(surf.Center.x, surf.Center.y, surf.Center.z, surf.MajorRadius, surf.MinorRadius, ((surf.MajorRadius - surf.MinorRadius)/2))
```

**buildShape in GEOReverse fails for building a Torus**
```python
def buildShape(self,boundBox):
        center, axis, Ra, Rb, Rc  = self.params  #Ra distance from torus axis; R radius of toroidal-cylinder
        if (abs(Rb-Rc)< 1e-5) and Ra > 0:
           self.shape = Part.makeTorus(Ra,Rb,center,axis)    # FreeCAD circular Torus
        else:
           self.shape = makeEllipticTorus(Ra,Rb,Rc,center,axis)      # Home made elliptic Torus 
 ```
There are two problems here. First, in `XMLinput.py` torus only has 4 parameters:
```python
params = ( p,v,Ra,R)
 ```
But in `buildShape` (see above) self.params is supposed to be of length 5.

The second problem is that it seems like the `Part` Libary doesn't have any `makeTorus()` function. At least I didn't find it in the docs.

vwbus1967, 2024-03-07:
Thank you for advise.  
I will try to check the other output files, too.

May I ask?  As my understanding, the torus is only supported now, I mean the elliptic torus is not able to convert.  Therefore, the r1=r2 is set.  I'm just complicated.  Am I correct?

According to the equation that the TX, TY, and TZ obey, a = MajorRadius, b = sub-MinorRadius, c = sub-MajorRadius.
b = k*c for the elliptic cross-section torus and b = c for circular. So I think the last article might not be taken the average of the difference a and b.

coeffs ='x0={},y0={},z0={},a={},b={},c={}'.format(surf.Center.x, surf.Center.y, surf.Center.z, surf.MajorRadius, surf.MinorRadius, ((surf.MajorRadius - surf.MinorRadius)/2))

I've just concentrated on the GEOUNED so far, I will try to understand the script written for GEOReverse.
Really helpful and informative.  thank you.

mxmws, 2024-03-08:
You are right about the formula. So the line should look like this, if I'm not mistaken:

```
coeffs ='x0={},y0={},z0={},a={},b={},c={}'.format(surf.Center.x, surf.Center.y, surf.Center.z, surf.MajorRadius, surf.MinorRadius,  surf.MinorRadius)
```

vwbus1967, 2024-03-10:
## OpenMCSurface issue for writing torus to py-file
```
coeffs ='x0={},y0={},z0={},a={},b={},c={}'.format(surf.Center.x, surf.Center.y, surf.Center.z, surf.MajorRadius, surf.MinorRadius,  surf.MinorRadius)
```
Yes.  I think so, too.  But I'm not an openMC user, so I'm not able to check for sure, unfortunately.

## buildShape in GEOReverse fails for building a torus
This main problem would be  missing import modules, FreeCAD and Part, in GEOReverse/Modules/splitFunction.py
I have checked that I can export a torus.stp using /testing/MCNP_Reverse/torus.mcnp with GEOReverse.

Adding
```
import FreeCAD, Part
```
in below
<https://github.com/GEOUNED-code/GEOUNED/blob/734f29a162cc4356e9d8f698d987dbfd6a709de6/src/GEOReverse/Modules/splitFunction.py#L1-L5>

### For the torus description in XMLinput.py,
> First, in XMLinput.py torus only has 4 parameters:
> `params = ( p,v,Ra,R)`

Paying attention to the following,
<https://github.com/GEOUNED-code/GEOUNED/blob/734f29a162cc4356e9d8f698d987dbfd6a709de6/src/GEOReverse/Modules/XMLinput.py#L361>
I would propose to set the same R for Rc to solve like;
```
params = ( p,v,Ra,R,R)
```
Because MCNPinput.py and XMLinput.py share the same `def buildShape` of `class Torus:` in Objects.py
<https://github.com/GEOUNED-code/GEOUNED/blob/734f29a162cc4356e9d8f698d987dbfd6a709de6/src/GEOReverse/Modules/Objects.py#L593-L598>

### The second problem
> the Part Libary doesn't have any makeTorus() function

As you say, I also couldn't find the makeTorus() in any module files.
But I don't think we need to worry.  The method is defined in FreeCAD.  [Here is a reference.](https://wiki.freecad.org/Topological_data_scripting)
The basic method like makeTorus() seems not to appear in the Mod folder as a .py module file.
If you worry about, please open FreeCAD python console and type 'Part.makeTorus' to check.
In the middle of typing, you can also find this method in popped up available complement list. 
(please see attached file) 
[image]

psauvan, 2024-03-13:
> Dear all,
>
> I'm trying to convert buried pipes and I'm facing to some conversion failures. Would you check and give some advices?
>
> The failures seem to relate to a torus conversion. To properly [...]

Yes, there were bugs in the torus conversion. Now it is solved, updating your code with the dev branch should solve the problem.
Concerning the conversion of your "BaseSample.step" solid, to translate correctly you must slice you solid in 3 parts  isolating the 3 buried pipes. Translating one by one GEOUNED is able to do the work correctly.  I attach an example of the slice.
[newbase.zip](https://github.com/GEOUNED-code/GEOUNED/files/14588969/newbase.zip)

Regarding the parameters of the Torus, the implementation is correct. 
In GEOUNED only torus with circular section can be translated because FreeCAD is unable to identify a torus with an elliptical section and as a consequence we cannot get the parameters of an elliptical torus.
In GEOReverse, elliptic torus are allowed. You have highlighted some bugs (calling FreeCAD and Part modules, or reading parameters from OpenMC  input ) that I will correct shortly.

closed 2024-03-13 by psauvan