GEOUNED-org/GEOUNED #36: Creating multiple instances of solids

Teancum-Q, 2024-04-02:
I recently updated to the latest version, and I'm experiencing a duplication in my solids. You can see what happens in the attached files. I'm not sure what is causing this. The 17 modeled solids become 34 solids with 17 perfectly overlaying the other 17.
[New folder.zip](https://github.com/GEOUNED-code/GEOUNED/files/14844105/New.folder.zip)

psauvan, 2024-04-09:
Your solids are repeated in the converted file because in the step file opened with FreeCAD all your solids are defined twice,

closed 2024-04-09 by psauvan

Teancum-Q, 2024-04-09:
For anyone else maybe new to freecad (21.2) , this is due to using select all for stp. export. The body, and its underlying pad are both selected if you select all in the tree or in 'edit' 'select all'. This creates two instances in the stp file of each object. Use bounding box (shift-b) in 3d view,  ctrl-left click each body you wish to export and not it's associated sub structure (pad, cut, etc). You could also do the reverse, but that sounds tedious.  Hope this helps.