GEOUNED-org/GEOUNED #46: [Feature request] - Continious intergration testing with Pytest

shimwell, 2024-04-12:
There is a nice opportunity to add some CI to this repo and test the code as it gets updated.

A non exhaustive list of things that could be tested

- [ ] installation routes to check the software can always be installed
- [ ] conversion of CAD to mcnp and serpent
- [ ] public CI based simulation is not possible for mcnp and serpent due to license but we could do still add the pytests and make the local testing easier for devs
- [ ] compare mcnp and serpent files with previously made mcnp files that we know work in simulation (regression test)
- [ ] conversion of CAD to openmc
- [ ] simulation of openmc files and check result with manually made geometry that matches the CAD or DAGMC geometry ([perhaps similar to model benchmark zoo](https://github.com/fusion-energy/model_benchmark_zoo/))
- [ ] test different python versions
- [ ] test use with different operating systems
- [ ] test with different methods of installing and connecting with freecad
- [ ] unit tests for functionality of code
- [ ] regression tests to make sure we still get the same outputs

I'm keen to contribute the above todo list if allowed.

Just raising an issue to check if others are keen on seeing this testing added?

I've made a start over [here](https://github.com/shimwell/GEOUNED/pull/3) if anyone is interested

shimwell, 2024-04-23:
Perhaps this package is useful for the performance testing
https://github.com/ionelmc/pytest-benchmark

alberto743, 2024-04-29:
In my humble opinion, software performance shall not be put as a priority.
As Donald Knuth says, "premature optimization is the root of all evil".

More specifically, we need instead to have object persistence (e.g. via `pickle`) to let the user interactively save the state of partially converted models.
At the same time, the correctness of the conversions shall be verified against specific unit and integral test cases, to have as a top priority the reliability of the results.
Continuous integration pipelines should trigger a battery of fast test cases capable of covering the features of the software.

alexvalentine94, 2024-06-14:
Closing as mostly completed

closed 2024-06-14 by alexvalentine94