GEOUNED-org/GEOUNED #55: [BUG] - volume mismatch between CAD and resulting CSG for two torus shapes

shimwell, 2024-04-18:
# Describe the bug
I've been converting a couple of files in the testing/inputSTEP folder and have got differences between the cell volume in CSG and CAD geometries.

|           | CSG volume | CAD volume |
|-----------|------------|------------|
| [face2.stp](https://github.com/GEOUNED-org/GEOUNED/blob/main/testing/inputSTEP/Torus/face2.stp) | 12672      | 9019       |
| [rueda.stp](https://github.com/GEOUNED-org/GEOUNED/blob/main/testing/inputSTEP/Torus/rueda.stp) | 344        | 360        |

# To Reproduce
Steps to reproduce the behavior I made this script that runs freecad to get the cad volumes and openmc to get the csg volume. 
```python
import math
try:
    import freecad  # importing conda package if present
except:
    pass
import Part
import pytest
from FreeCAD import Import
import openmc
from pathlib import Path
import GEOUNED

input_step_file = Path('testing/inputSTEP/Torus/face2.stp') # cad_vol=9019,  csg vol=12672
# input_step_file = Path('testing/inputSTEP/Torus/rueda.stp') # cad vol=360, csg vol=344

template = (
    "[Files]\n"
    "title = 'Input Test'\n"
    f"stepFile = {input_step_file.resolve()}\n"
    f"geometryName = {input_step_file.stem}\n"
    "outFormat = ('openMC_XML')\n"
    "[Parameters]\n"
    "compSolids = False\n"
    "volCARD = False\n"
    "volSDEF = True\n"
    "voidGen = True\n"
    "dummyMat = True\n"
    "minVoidSize = 100\n"
    "cellSummaryFile = False\n"
    "cellCommentFile = False\n"
    "debug = False\n"
    "simplify = full\n"
    "[Options]\n"
    "forceCylinder = False\n"
    "splitTolerance = 0\n"
    "newSplitPlane = True\n"
    "nPlaneReverse = 0\n"
)

with open("config.ini", mode="w") as file:
    file.write(template)

GEO = GEOUNED.GEOUNED('config.ini')
GEO.SetOptions()
GEO.outFormat = ("openMC_XML")
GEO.Start()

Import.insert(str(input_step_file), "converted-cad")
result = Part.Shape()
result.read(str(input_step_file))

openmc_xml_file=Path(input_step_file.stem).with_suffix('.xml')
materials = openmc.Materials()
geometry = openmc.Geometry.from_xml(path=openmc_xml_file, materials=materials)
cells = geometry.get_all_cells()

bb = result.BoundBox
llx, lly, llz, urx, ury, urz = bb.XMin, bb.YMin, bb.ZMin, bb.XMax, bb.YMax, bb.ZMax
llx, lly, llz, urx, ury, urz = llx/10, lly/10, llz/10, urx/10, ury/10, urz/10

cell_vol_calc = openmc.VolumeCalculation(
    domains=[cell for _, cell in cells.items()],
    samples=40_000_000,
    lower_left=(llx, lly, llz),
    upper_right=(urx, ury, urz),
)

settings = openmc.Settings()
settings.volume_calculations = cell_vol_calc
settings.run_mode = "volume"
model = openmc.Model(geometry, materials, settings)
model.run()

vol_calc_file = openmc_xml_file.with_name(f"volume_1.h5")
cell_vol_calc_results = openmc.VolumeCalculation.from_hdf5(vol_calc_file)

for i, solid in enumerate(result.Solids, start=1):
    volume_of_csg_cell = cell_vol_calc_results.volumes[i].nominal_value
    volume_of_csg_cell_std_dev = cell_vol_calc_results.volumes[i].std_dev
    # converts from mm3 in cad to cm3 in csg
    volume_of_cad_cell = solid.Volume * 0.001
    print(f'volume_of_cad_cell {volume_of_cad_cell} volume_of_csg_cell {volume_of_csg_cell} std_dev {volume_of_csg_cell_std_dev}')
    assert math.isclose(volume_of_cad_cell, volume_of_csg_cell, rel_tol=0.05)
```

**IMPORTANT**
If you failed to convert a specific geometry, please isolate the geometry features producing the bug (ideally a single body), save it as a STEP file and attach it to this issue. Include also all the files used in the conversion as well as the GEOUNED version used.
>>> The two step files are single body geometries

# Expected behavior
volumes of csg and cad should match or be closer to matching

# Please complete the following information):
 - OS: [e.g. Linux or Windows] - ubuntu 22.04
 - GEOUNED version - dev branch
 - Python version
 -  python  3.11 conda. The script runs in a conda env where both freecad and openmc have been conda installed.

psauvan, 2024-04-19:
The problem has been fixed.
In the case of the "rueda.stp" model there was a bug in the code. Additional plane was wrongly added in the cell definition. This bug is now fixed.
Concerning the "face2.stp" model, the translation is correct. I have check the volume using MCNP simulation and ray tracing volume sampling using a spherical source.
For stochastic volume evaluation, sampling using ray tracing with spherical source is the most efficient.
I don't know what method is using openMC in his method "VolumeCalculation". I think it would be fine if a spherical source like the MCNP one's could be defined with openMC and be used to evaluate volumes with ray tracing approach.

closed 2024-04-19 by psauvan

shimwell, 2024-04-19:
Many thanks for fixing. I shall increase the particle count for face2.stp and hopefully get the right answer