GEOUNED-org/GEOUNED #70: Reduce the testing on the dev branch

AlvaroCubi, 2024-04-23:
Currently a PR into the dev branch triggers the whole CI testing suite which is too slow.

It would be better to run the whole suit only when merging to main. Maybe we could still run some tests that would finish in less than a couple of minutes.

psauvan, 2024-04-25:
Yes, I think we have to rethink the tests on PR.  We cannot wait so long test running before approving a PR. If we have  to approve various PR this would not be efficient and very time and resources consuming. 
May be for the PR branch we can :
 - reduce the tests only to 1 S.O. (Ubuntu for instance)
 - ask people submitting a PR to run the test on their own before making the PR.
 - set the test as an option for the approver of the PR to run it or not. (sometimes it is evident that PR will not change the code behaviour)

alberto743, 2024-04-29:
In addition to what @psauvan said, maybe we should also classify the test cases by means of `pytest.mark`.
This would allow deciding the subset of the whole suite that should be run in preparation of a pull request.

alexvalentine94, 2024-06-14:
As per @shimwell comment, testing reduced.

closed 2024-06-14 by alexvalentine94