GEOUNED-org/GEOUNED #77: [BUG] - Issue with converting bodies containing tori with 'forceCylinder' set to 'False'

teade, 2024-04-24:
I have been having problems converting CAD containing a number of toridal surfaces with 'forceCylinder' set to 'False'. You can see from the CAD and visit images below that some space around the toroidal surfaces is being not being defined as part of the solid body:

[image]
[image]

Looking at the code it appears that this is caused by the incorrect definition of the cone surfaces added to the cell definition to limit the extent of the toriodal surfaces. As cone surfaces have two 'sheets' there is a section of the code which selects the relevant 'sheet'. I think there is an error in the logic of this line:

https://github.com/GEOUNED-org/GEOUNED/blob/main/src/geouned/GEOUNED/Conversion/CellDefinition.py#L601
```python 
        ConeAxis = axis if za < 0 else -axis
```

I am not entirely sure what 'za' physically is. However changing this line to check which side (positive or negative) of the apex, the height of the first Vparams point is seems to fix the issue. I will raise a PR, but changing the line to:

```python
        ConeAxis = axis if (face.Surface.Center.dot(axis) + z1 - Apex.dot(axis)) > 0.0 else -axis
``` 
gives:

[image]

Note that there is also an issue with this geometry when 'forceCylinder' is set to 'True'. However, I will raise a seperate issue for this as it will need a different fix.

psauvan, 2024-04-24:
Could you please attach the stp solid producing the error?
I would like to know why in this case the axis is not oriented correctly.
the za variable is the algebraic distance from torus center to the cone apex, taking as positive value the torus axis direction.

teade, 2024-04-24:
I attach the stp file. 

[example_tori.zip](https://github.com/GEOUNED-org/GEOUNED/files/15095762/example_tori.zip)

I think the problem is that you cannot determine the axis (which in turn determines the 'sheet') from just the distance between the apex and the torus centre. You could have a cone which has apex above or below the centre of the torus and still requires you to use the negative sheet.

You need to know which side (positive or negative) of the apex, the point you are trying to intersect lies - this should tell you which axis/sheet you need to use.

psauvan, 2024-04-24:
Thank you!
Yes, there were a bug with the test of the axis direction.
Can you make the small change I propose in your PR so I can approve it.

commit bc74e7ef (2024-04-25): fixes #77 (#78)

* Fix to additional cone surface when defining tori

* Updated fix as per psauvan suggestion
closed 2024-04-25 by psauvan