GEOUNED-org/GEOUNED #88: [BUG] - conversions inherit previous conversion attributes

shimwell, 2024-04-25:
# Describe the bug

When running more than one conversion in the same python script the attributes of Options, Tolerances and Numerical_format are carried between conversions.

This is difficult for the user to spot and could lead to a decent amount of confusion.

For example if the first conversion changes any of these setting in their config.ini file and does not over write them with the next config file then these new settings be carried between conversions. Impacts this settings in the config
```
[Options]
forceCylinder
newSplitPlane
delLastNumber
verbose
enlargeBox
nPlaneReverse
splitTolerance
scaleUp
quadricPY
Facets
[Tolerances]
prnt3PPlane
relativeTol
relativePrecision
value
distance
angle
pln_distance
pln_angle
cyl_distance
cyl_angle
sph_distance
kne_distance
kne_angle
tor_distance
tor_angle
min_area
[MCNP_numeric_format]
P_abc
P_d
P_xyz
S_r
S_xyz
C_r
C_xyz
K_xyz
K_tan2
T_r
T_xyz
GQ_1to6
GQ_7to9
GQ_10
```

# To Reproduce
this example converts a stp file with a set relativePrecision  then converts a second step file without specifying the relativePrecision and it would end up not having the default Tolerance.relativePrecision  but instead it would get the relativePrecision setting from the previous conversion
```python
from pathlib import Path
from geouned import GEOUNED
from geouned import Tolerances

print(Tolerances.relativePrecision)
#  prints to terminal the default value of relativePrecision
#  which is set in here in the code https://github.com/GEOUNED-org/GEOUNED/blob/bc74e7ef28a50290fdfd3847cea7c1956fea3418/src/geouned/GEOUNED/Utils/Options/__init__.py#L18
# >>> 1e-06

# finds a cad file to convert
path_to_cad = Path("testing/inputSTEP")
step_files = list(path_to_cad.rglob("*.stp")) + list(path_to_cad.rglob("*.step"))
input_step_file = step_files[2]

# creates the config file contents
template = (
    "[Files]\n"
    "title = 'Input Test'\n"
    f"stepFile = {input_step_file}\n"
    f"geometryName = '1st_conversion\n"
    "outFormat = ('mcnp', 'openMC_XML')\n"
    "[Parameters]\n"
    "compSolids = False\n"
    "volCARD = False\n"
    "volSDEF = True\n"
    "voidGen = True\n"
    "dummyMat = True\n"
    "minVoidSize = 100\n"
    "cellSummaryFile = False\n"
    "cellCommentFile = False\n"
    "debug = False\n"
    "simplify = no\n"
    "[Options]\n"
    "forceCylinder = False\n"
    "splitTolerance = 0\n"
    "newSplitPlane = True\n"
    "nPlaneReverse = 0\n"
    "[Tolerances]\n"
    "relativePrecision = 2e-6"  # here we make u new number for relativePrecision
)

with open("config1.ini", mode="w") as file:
    file.write(template)

GEO = GEOUNED('config1.ini')
GEO.SetOptions()  # this changes relativePrecision to our new number 2e-6
GEO.Start()

# Tolerances stays set as the new number
print(f'Tolerances.relativePrecision= {Tolerances.relativePrecision}')

# now if we want to do another conversion
# this time we don't state the tolerances in the config
template = (
    "[Files]\n"
    "title = 'Input Test'\n"
    f"stepFile = {input_step_file}\n"
    f"geometryName = '2nd_conversion'\n"
    "outFormat = ('mcnp', 'openMC_XML')\n"
    "[Parameters]\n"
    "compSolids = False\n"
    "volCARD = False\n"
    "volSDEF = True\n"
    "voidGen = True\n"
    "dummyMat = True\n"
    "minVoidSize = 100\n"
    "cellSummaryFile = False\n"
    "cellCommentFile = False\n"
    "debug = False\n"
    "simplify = no\n"
    "[Options]\n"
    "forceCylinder = False\n"
    "splitTolerance = 0\n"
    "newSplitPlane = True\n"
    "nPlaneReverse = 0\n"
)

with open("config2.ini", mode="w") as file:
    file.write(template)

GEO = GEOUNED('config2.ini')
GEO.SetOptions()  # this does not change the relativePrecision back to the default
GEO.Start()

# Tolerances stays set as the new number and the 2nd conversion uses the tolerance of the first example
print(f'Tolerances.relativePrecision= {Tolerances.relativePrecision}')
```

# Expected behavior
When the user makes a new GEOUNED instance then it should come with the defaults, not with the previous conversion values.

# Error message
no error message

# Please complete the following information):
 - OS: linuix
 - GEOUNED dev
 - Python version 3.10

psauvan, 2024-04-26:
This is because "Options" and "Tolerances" parameters are stored in a class as class attributes. During the initialization the set method changes the attribute value of the classes "Options" and "Tolerances" for the attribute name defined in the user template. If in the second call of the GEOUNED instance the attributes changed in the first instance call are not defined as input data in the second template, their value doesn't change from the first assignation because, the classes "Options" or "Tolerances" are the same as in the first call of the GEOUNED instance (after updating the value of their attibute).
The solution would be to create a new "Options" and "Tolerances" instances each time the GEOUNED instance is called or delete these classes if there exits when GEOUNED instance is called.
P.D.: I have defined the parameters of the options and tolerance through classes and class attibutes to have access to their values in the code as global variables (I don't know if it is the best way to do this)

psauvan, 2024-04-28:
The proposed PR #102 is doing more or less what your are proposing.

Originally I separated the input parameters in four groups (as they are now) depending on the type of parameters and their use for the translation process. 
The first group which is the one correspond to attributes passed directly to the class ("stepFile","outFormat",....) are the basic inputs that can change on each different run (same model to translate or different one).
The other groups defined as classes are not supposed to be changed from one simulation to other, they are here only if in specific case the translation is not good right and intermediate/advanced user want to change some parameter to solve the issue. The parameters passed through the class may have been hard coded,

Returning to your proposal, I am agree to pass the Options, Tolerances, and McnpNumericFormat as local classes to each CadToCSG instance. But formerly, when I tryied to do this, I face the problem that there were several method calls between the point where the class were defined and the methods/functions using their attributes, and passing the information through the different calls wasn't easy nor elegant. That's why I decided to use global classes. 

I think this work should be included as part of the refactoring work, and cannot be performed easily as isolated task.

psauvan, 2024-04-28:
I didn't understand what method like ```CadToCsg.export_mcnp(startCell=2)``` pretend to do.