GEOUNED-org/GEOUNED #94: [BUG] - Issue with converting bodies containing tori with 'forceCylinder' set to 'True'

teade, 2024-04-26:
# Describe the bug
There is an issue converting toridal surfaces with 'forceCylinders' set to 'True'. See images of CAD and conversion below. 

[image]
[image]

This seems to be due to the radius of the cylinder limiting the extent of the toroidal surface being set to the minimum of d1 or d2.

# To Reproduce
Steps to reproduce the behavior:
for example:
1. Use attached stp file
2. Use current main branch commit: https://github.com/GEOUNED-org/GEOUNED/commit/bc74e7ef28a50290fdfd3847cea7c1956fea3418
3. Ensure that 'forceCylinders' option is set to True (this seems to be the default)

# Expected behavior
This is the geometry you would expect to see. 

[image]

It appears that when inSurf is true the maximum value of d1 or d2 is required.

psauvan, 2024-04-26:
Perhaps, it is better to remove the "forceCylinder" option and use only cones to define the annex torii surfaces.
What do you think? Do you see any reason to keep cylinders as annex surface for torus (of course except in the case in which the cone parameters degenerate to a cylinder)?

alexvalentine94, 2024-06-14:
New issue to be raised on latest comments.

PR #96 (2024-04-26): Fix to position of additional cylindrical surface used to limit torus extent
This PR fixes #94. Instead of always using the minimum radius of d1 and d2, if the cell is inside of the surface it uses the maximum radius in order to ensure we do not clip the cell unnecessarily.
closed 2024-06-14 by alexvalentine94