GEOUNED-org/GEOUNED #132: [BUG] - geouned.py script not working straight out of installation

AlvaroCubi, 2024-05-07:
# Describe the bug
When a new user clones the repository, the geouned.py script will not work (for both main and dev branches).

The issues vary depending on the branch. It may be due to the use of lower case and upper case. Also if the script is named geouned.py which is the same name as the module geouned, it may produce more issues.

# To Reproduce
Steps to reproduce the behavior:
for example:
1. Make a clean clone of the repository
2. Modify the geouned.py so it points to the correct paths
3. See that it doesn't work (running with the Python that comes with FreeCAD)

# Expected behavior
The script should be able to be used as the main entry point for the user to the program.

# Please complete the following information):
 - OS: Windows
 - GEOUNED version: main 76ef697

# Proposal
We may want to have tests to check that the scripts to interact with geouned still work after changes are made.

AlvaroCubi, 2024-05-08:
@shimwell yes thank you very much, I got it working by changing the script name and tweaking the contents, I think we should keep the issue open until the problem is fixed at least in main

alexvalentine94, 2024-06-14:
Assume closed.

closed 2024-06-14 by alexvalentine94