GEOUNED-org/GEOUNED #146: [BUG] - AttributeError: 'int' object has no attribute 'check'

evvaletov, 2024-05-11:
# Describe the bug
I ran conversion to MCNP file for a STEP file. With "simplify = no", the process completes without errors. However, with "simplify = full" or "simplify = void" I get the following error `AttributeError: 'int' object has no attribute 'check'`.

# To Reproduce
Steps to reproduce the behaviour:
The issue occurred at the stage of simplification of void cells, which potentially complicates isolating the bug in a complex geometry. I performed some debugging on my end, and it seems that cell definitions like the following may be problematic:

3450  99    -1.2000e-03 14666 -14706 14719 -14649 14658 -14693 (-11968:(-14411:
           -14410:-11975 14775:-11973:-11966:11951:11972:-11964) (-14415:-14404:
           11957:-14419:-11974:-11972:11964)) ((-14427:-14426:11976:-14419:
           -11974:11978) (-14435:-14434:-14429:14776 -11987:-11982:-11979:11951:
           -11978):-11981:-11985) (-12073:(11636:11996:12066) (-11993:11992:
           -11996) (10859:-12065:-12066)) (12000:-11988:-11990:-11999) (-12024:
           12023:-12020:12026) (12018:-12013 (12022:12026:-12024:12021)) (11477:
           12026:-12023) (-11993:11992 (12001:-11999:11988)) (-11999:(12000:
           12007:-12006) (12006:12001) (-14778:12014:12009:-12007)) -12012 
           -12011 -11994 (-11998:11997:(11989:-11988) (-12006:12008)) (12002:
           -14777:11991:11990:-11999) (12025:-12024:12020:-12021) (-10201:11803:
           11802:-11636) (-11485:11484:11476:-11475:-11477)
           $'Air assigned to Void'
           $Automatic Generated Void Cell. Enclosure(121.617, 134.735, 870.012, 882.961, -20.817, -7.649)
           $Enclosed cells : (1270, 1319, 1342, 1343, 1344, 1345, 1346, 1350)

I am not certain but it could be that check() was attempted to be called for 12012 in this specific cell.

# Expected behavior
The simplification of void cells completes without errors.

# Error message

```
build complementary 13 231
Traceback (most recent call last):
  File "convert.py", line 68, in <module>
    GEO.Start()
  File "C:\Users\Eremey\.conda\envs\geouned\lib\site-packages\geouned\GEOUNED\__init__.py", line 550, in Start
    MetaVoid = Void.voidGeneration(
  File "C:\Users\Eremey\.conda\envs\geouned\lib\site-packages\geouned\GEOUNED\Void\Void.py", line 43, in voidGeneration
    voids = GetVoidDef(newMetaList, Surfaces, EnclosureBox, setting, Lev0=True)
  File "C:\Users\Eremey\.conda\envs\geouned\lib\site-packages\geouned\GEOUNED\Void\Void.py", line 120, in GetVoidDef
    cell, CellIn = z.getVoidComplementary(Surfaces, simplify=simplifyVoid)
  File "C:\Users\Eremey\.conda\envs\geouned\lib\site-packages\geouned\GEOUNED\Void\VoidBoxClass.py", line 266, in getVoidComplementary
    newSolid = removeExtraSurfaces(solDef, CTable)
  File "C:\Users\Eremey\.conda\envs\geouned\lib\site-packages\geouned\GEOUNED\Utils\BooleanSolids.py", line 334, in removeExtraSurfaces
    subCell.check()
AttributeError: 'int' object has no attribute 'check'
```

# Screenshots
N/A

# Please complete the following information):
 - OS: Oracle Linux 8
 - GEOUNED version: 0.21.2
 - Python version: 3.8.19

# Additional context
N/A

evvaletov, 2024-05-13:
Hi @shimwell , it is on the main branch, as I used the command 
`pip install git+https://github.com/GEOUNED-code/GEOUNED.git`
for installation.

evvaletov, 2024-05-15:
I tried installing GEOUNED from the dev branch and got the same error. I sent @psauvan a copy of the STEP file.

alexvalentine94, 2024-06-14:
solved

closed 2024-06-14 by alexvalentine94