GEOUNED-org/GEOUNED #164: [BUG]step to mcnp using simplify = full raise error but simplify = no works.

WalonMay, 2024-05-15:
# Describe the bug
1. step to mcnp using simplify = full raise error but simplify = no works.
2. result that simplify = no seems weired(cell 3-35 is same and have strange surface 0).

# To Reproduce
Steps to reproduce the behavior:
for example:
1. use config[config.ini.txt](https://github.com/GEOUNED-org/GEOUNED/files/15319378/config.ini.txt)
2. get weired result[result.mcnp.txt](https://github.com/GEOUNED-org/GEOUNED/files/15319404/result.mcnp.txt)
3. change simplify = full, raise error
"""
simplify cell 1
Traceback (most recent call last):
  File "C:\Users\Administrator\Desktop\test\converter.py", line 57, in <module>
    GEO.Start()
  File "C:\Users\Administrator\Desktop\ph1\FreeCAD\bin\lib\site-packages\geouned\GEOUNED\__init__.py", line 566, in Start
    CT = buildCTableFromSolids(Box, (c.Surfaces, Surfs), option="full")
  File "C:\Users\Administrator\Desktop\ph1\FreeCAD\bin\lib\site-packages\geouned\GEOUNED\Utils\BooleanSolids.py", line 253, in buildCTableFromSolids
    if type(surfaces[surfaceList[0]]) is GEOUNED_Surface:
KeyError: 0
"""

# Please complete the following information):
 - OS: windows 10
 - GEOUNED version 1.0.1
 - Python version 3.8.1(FreeCAD v0.21)

# Suggestion
after remove bad surface, can we check the cell and see if it is valid

psauvan, 2024-05-15:
The surface 0 in the cell definition is weird, but the repetition of the same cell 3 to 35 is weird too.
Have you tried to open you stp file with FreeCAD? 
If you can share your stp file, we can look at the reason of the problem.

WalonMay, 2024-05-15:
Here is the file[Crackled tile 100x100x9mm.stp.txt](https://github.com/GEOUNED-org/GEOUNED/files/15323490/Crackled.tile.100x100x9mm.stp.txt), download from https://grabcad.com/library/crackled-tile-1, thank you for your immediate response.

psauvan, 2024-05-15:
I am very sorry but GEOUNED is not designed to translate CAD with such details and complexity. I would be glad if in a future GEOUNED could do it but currently it is impossible.

WalonMay, 2024-05-16:
hope geouned do it better, also.
by the way, may i ask if i could ignore the cracked in the board like [steel_cracked.stp.txt](https://github.com/GEOUNED-org/GEOUNED/files/15328608/steel_cracked.stp.txt) and finally get the 6-face(as less as possible) cell card by setting the .ini file, maybe some models have too much surfaces and need more simplify.

closed 2024-05-29 by WalonMay