GEOUNED-org/GEOUNED #187: [Feature request] - Make the automatic addition of comments to cells optional

AlvaroCubi, 2024-05-21:
# Is your feature request related to a problem? Please describe.
Some workflows do not require the automatic addition of comments to the MCNP cells. Moreover, these comments may provide no useful information and just clutter the model.

# Describe the solution you'd like
I would like to have a parameter in the config file that is called something like: "cell_comments" which would default to True but if set to False: no comments on the MCNP cells will be written.

AlvaroCubi, 2024-05-21:
> perhaps another way to solve this is to have separate export methods for each MC code

That would be fine with me

KBGrammer, 2024-06-05:
Somehow my comments are coming up as empty regardless. I'm trying to convert with debug on and it seems that logger.info(m.Comments) is empty in decompose_solids.

Any ideas how that would happen and how to remedy it?