GEOUNED-org/GEOUNED #222: [Feature request] - Add metadata/archiving feature

alexvalentine94, 2024-06-14:
# Is your feature request related to a problem? Please describe.
Enable a way of having full traceability for the input parameters used to create GEOUNED models

# Describe the solution you'd like
A method for outputting the input parameters and additional metadata describing the translation e.g. geouned version, time date. Could be output as JSON.

# Describe alternatives you've considered
-

# Additional context
-

alberto743, 2024-06-14:
Having the configuration serialization will mean having a part of the persistent objects mentioned in #114

alberto743, 2024-06-14:
To implement this feature, it would be convenient to convert `Options`, `Tolerances`, `NumericFormat`, `Settings` to [dataclasses](https://docs.python.org/3/library/dataclasses.html).