GEOUNED-org/GEOUNED #227: [Feature request] - Material assignment by colour

alexvalentine94, 2024-06-14:
# Is your feature request related to a problem? Please describe.
Enabling different workflows whereby the CAD is coloured according to differnt materials. 

# Describe the solution you'd like
Currently the materials are assigned based on the naming in the structure tree. In the part of the code where the materials are assigned we can simply use the rgb values from the step files. 

# Describe alternatives you've considered
Keep as present. 

# Additional context
-

KBGrammer, 2024-06-17:
The step file standard allows for a material specification, including a density. Is this something that can be easily pulled from meta data?

psauvan, 2024-06-18:
> The step file standard allows for a material specification, including a density. Is this something that can be easily pulled from meta data?

What do you mean by meta data?
The information on the density scaling factor is available in the output file "(geometryName)_Summary.txt". In this file the density value is the nominal density written in the ```materials.txt``` file multiplied by the scaling factor introduced in the CAD file for each solid.

KBGrammer, 2024-06-18:
The step file standard supports a material name and density for each part, which I thought that GEOUNED doesn't currently use. The problem is that it's often omitted by CAD software during export. I don't have an example file that contains that information to play around with.