GEOUNED-org/GEOUNED #228: [Feature request] - Option to scale geometry

alexvalentine94, 2024-06-14:
# Is your feature request related to a problem? Please describe.
Often we encounter issues related to coincident surfaces in MCNP when using universes. 

# Describe the solution you'd like
One way around this is to nudge or scale the geometry by a very small amount to avoid coincident surfaces. This would be a nice feature option in GEOUNED. 

# Describe alternatives you've considered
-

# Additional context
-

alexvalentine94, 2024-06-14:
As discussed with @psauvan, default in the code currently is factor 10 for mm - cm. User input option can be added to allow flow specification of scaling factor which defaults to 1.

psauvan, 2024-06-15:
Scaling is performed when surface parameters are written in the output file.
The function below writes the surface parameters to MCNP output file.
 https://github.com/GEOUNED-org/GEOUNED/blob/ec9a61b2dda6011b594228a9974843741643ccac/src/geouned/GEOUNED/write/functions.py#L241
In this function (and corresponding functions writing other output formats), the way the factor is applied is not consistent between the diferent surface type. Sometimes the parameter value is scaled, 
```python
        Pos = surf.Center * 0.1
        rad = surf.Radius * 0.1
```
sometimes only the scaled value in written in the output file.
```python
                mcnp_def = "{:<6d} PX  {:{x}}".format(id, D / 10.0, x=numeric_format.P_xyz)
```

When the additional scaling factor is implemented, a homogenezation of the the way the parameters are scaled should be carried out.