GEOUNED-org/GEOUNED #252: [BUG] - Cannot install from conda

dodu94, 2024-07-03:
# Describe the bug
I tried to install geouned from conda on a fresh enviroment (python 3.11) without success. Error can be seen in the screenshot.

# To Reproduce
Steps to reproduce the behavior:
for example:
1. create a fresh conda environment
2. follow installation guide for conda and windows
3. See error

# Screenshots
[image]

# Please complete the following information):
 - OS: Windows
 - GEOUNED version: latest conda package?
 - Python version: 3.11

dodu94, 2024-07-03:
different PC, newer version of conda (22.X instead of an old 4.X) and I get a different error:

[image]

dodu94, 2024-07-04:
Is it worth to start a troubleshooting section in the doc?

SterlingButters, 2024-08-08:
Working for me in conda 23.3.1

akolsek, 2024-08-26:
@dodu94, whenever you can, please confirm if the error persists with the new release on conda-forge.

dodu94, 2024-08-26:
I still get the same error for conda 22.X (on my laptop). 

On conda 23.X (on my WS), instead, it took something like 20 mins if this endless looping:
```
Collecting package metadata (current_repodata.json): done
Solving environment: unsuccessful initial attempt using frozen solve. Retrying with flexible solve.
Solving environment: unsuccessful attempt using repodata from current_repodata.json, retrying with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: |
```

but finally it managed to install

dodu94, 2024-08-26:
Also please consider the fact that conda now moved to a mandatory paid plan for organizations of more than 200 employees (https://www.anaconda.com/pricing) so its popularity may fall a bit in the future (at least in pure open source environments)