GEOUNED-org/GEOUNED #258: [Feature request] - On-axis surface simplififcation

repositony, 2024-07-31:
# Is your feature request related to a problem? Please describe.
**This is not anything important, just a small quality of life thing** 

When something like a cylinder is on-axis MCNP will replace the surface and give a warning. For large models this can fill the logs with warnings and scare less experienced users into thinking there are issues with the conversion. 

[image]

# Describe the solution you'd like
This could be a small check while writing surfaces to MCNP cards. 

For example, if the precision set by the user is low then the output will just be `0.0` for a cylinder like this:

[image]

If, given the user precision, this is filled with zeros then it can be simplified to a `CZ` cylinder instead so that MCNP does not have to do it.  

[image]

# Describe alternatives you've considered
This could just be left as it is because MCNP will do this for you anyway. It was just a suggestion.

psauvan, 2024-08-02:
From my side I would let the surface definition like it is. Sometimes the rounding of the coefficients leads to geometry error, keeping the long surface definition would help the user to find the origin of the lost particle (e.g. a cylinder that seems to be parallel to an axis but actually not) .

What can be done is to introduce an option letting the user to choose between the current way of defining surface or simplifying the definition when rounding the parameters leads to zero coefficients.

repositony, 2024-08-03:
Ecellent point, I had not considered this. Anything that makes it more difficult for people to debug geometry issues is not worth having so I'd probably just leave it as-is then. I'll close the issue.

closed 2024-08-03 by repositony