GEOUNED-org/GEOUNED #260: [BUG] - Windows installation seems to be broken

AlvaroCubi, 2024-08-05:
# Describe the bug
Following the instructions for the Windows installation with Anaconda did not work for me. I created a new environment, activated it and used the command: conda install -c conda-forge geouned -y

# To Reproduce
Follow the instructions for installation on Windows via Anaconda as stated in the documentation.

# Expected behavior
A full installation that doesnt require the installation of FreeCAD independently.

# Error message
(using_geouned) D:\WORK\using_GEOUNED>conda install -c conda-forge geouned -y
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - geouned

Current channels:

  - https://conda.anaconda.org/conda-forge/win-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

# Please complete the following information):
 - OS: Windows
 - GEOUNED version: tried installing the 05/08/2024
 - 3.11 as described in the installation commands

AlvaroCubi, 2024-08-05:
Thank you @shimwell, the link seems to successfully install GEOUNED but not the FreeCAD software. I get the error:
ModuleNotFoundError: No module named 'FreeCAD'

jon-proximafusion, 2024-08-06:
As conda can install from the url but does not find the package when asked I think this is something with the locally conda. Perhaps try some of the solutions here
https://stackoverflow.com/questions/64651410/conda-cannot-find-packages-to-install

SterlingButters, 2024-08-09:
> Thank you @shimwell, the link seems to successfully install GEOUNED but not the FreeCAD software. I get the error: ModuleNotFoundError: No module named 'FreeCAD'

@AlvaroCubi, you should be able to just `conda install freecad` separately although I admittedly did not require this step

jon-proximafusion, 2024-08-09:
I notice we had a merge from Develop to main recently, so perhaps there will be a github release soon? which will generate a new to pypi distribution. Then we can update the conda package on conda forge. Perhaps this helps

akolsek, 2024-08-12:
@AlvaroCubi, would you be able to help with a new release?

akolsek, 2024-08-26:
@AlvaroCubi, whenever you can, please confirm if the error persists with the new release on conga-forge.

AlvaroCubi, 2024-09-20:
After upgrading Anaconda to the last version I have been able to install geouned + freecad following the windows instructions. It needed a long time (2+ hours). Thank you guys for the help!

closed 2024-09-20 by AlvaroCubi