GEOUNED-org/GEOUNED #269: [BUG] - cell references a null surface and contains other bounding surfaces.

SterlingButters, 2024-08-09:
# Describe the bug
A clear and concise description of what the bug is. Remember to try to reproduce your bug with the latest version of the _dev_ branch, the problem may already have been solved.

# To Reproduce
Steps to reproduce the behavior:
for example:
1. Generate `csg.mcnp` with default arguments
2. Observe null surface (`0`) in cell definitions

**IMPORTANT**
Cannot provide problem parts. 

# Expected behavior
Didn't expect any surfaces with ID `0` since `startSurf=1`

# Error message
MCNP:  `fatal error.  cell       <cell> references a null surface and contains other bounding surfaces.`

# Please complete the following information):
 - OS: Windows
 - GEOUNED version: 1.2.0
 - Python version: 3.11

psauvan, 2024-08-10:
It is very difficult to found the reasons of the bug with so little information. 
May be the CAD file has some features that FreeCAD doesn't handle correctly and produce this error in GEOUNED. 
Try to open your CAD file with FreeCAD and check if it looks like your original CAD.

SterlingButters, 2024-08-12:
@psauvan Done - not seeing anything weird in the representation of the geometry. I was hoping there was some obvious explanation for why GEOUNED might be introducing a null surface (ID=0) e.g. inappropriate indexing.  Is there a way for me to determine what surface in the geometry GOEUNED intended for the null surface (or is there another known reason for why GEOUNED would introduce a null surface)?

akolsek, 2024-08-12:
@SterlingButters, if you use SpaceClaim, try to check that your geometry doesn't have splines or that splines were not introduced while saving the geometry in a different format. I have seen a "surface ID 0" issue when a spline was present in the model. Removing the spline solved the issue.

SterlingButters, 2024-08-12:
@akolsek I *am* using SC so lemme check!

closed 2024-08-12 by SterlingButters