GEOUNED-org/GEOUNED #272: [BUG] - Installation error with conda

SeongHeeHong, 2024-08-12:
Hello. this is SeongHee Hong.
Now I trying installation GEOUNED with conda.
However i got a error massage like attached figure.
What can i do to solve this problem?

ERROR conda.core.link:_execute(950): An error occurred while installing package 'conda-forge::qt-5.15.8-h91493d7_0'.
Rolling back transaction: done

LinkError: post-link script failed for package conda-forge::qt-5.15.8-h91493d7_0
location of failed script: C:\Users\shhong\AppData\Local\miniconda3\envs\new_env\Scripts\.qt-post-link.bat
==> script messages <==
<None>
==> script output <==
stdout:         1개 파일이 복사되었습니다.

stderr: 'chcp'은(는) 내부 또는 외부 명령, 실행할 수 있는 프로그램, 또는
배치 파일이 아닙니다.
'chcp'은(는) 내부 또는 외부 명령, 실행할 수 있는 프로그램, 또는
배치 파일이 아닙니다.
'chcp'은(는) 내부 또는 외부 명령, 실행할 수 있는 프로그램, 또는
배치 파일이 아닙니다.

return code: 1

SeongHeeHong, 2024-08-12:
Yes, I did install below process.
conda create --name new_env python=3.11
conda activate new_env
conda install -c conda-forge geouned -y

akolsek, 2024-08-26:
@SeongHeeHong, can you confirm if the error persists? A new version of GEOUNED has been released on conda-forge.