GEOUNED-org/GEOUNED #275: [BUG] - GEOUNED creating a cell referencing surface 0 (S0) and not defining the surface

raska-s, 2024-08-22:
# Describe the bug
I am trying to convert a STEP file into an OpenMC .xml file. The conversion via GEOUNED succeeds but it creates cells that references a non-existent surface (surface 0). This causes an issue when loading in OpenMC, as it won't be able to locate the surface. No error is thrown by GEOUNED.
The STEP file and the OpenMC .xml and .py files are available in this [link to download.](https://drive.google.com/drive/folders/1ooadYzZ4MO0yjpsOjjzRCA07vtV3NdCQ?usp=drive_link)

# To Reproduce
Steps to reproduce the behaviour:
1. Download the Defeature_Simplify_0.stp file from the link above
2. Run the following code:
```
myOptions = geouned.Options(forceNoOverlap=False)
geo = geo = geouned.CadToCsg(options = myOptions)
geo.load_step_file(filename=input_file_path) # change input file path to the step file
geo.start()
geo.export_csg(title = 'title', geometryName='geometry_name', outFormat=("openmc_xml", "openmc_py"))
```
The code will produce OpenMC geometry .xml and .py files with cells that reference a non-existent surface S0. This can be shown in the below snippet of the first cell's definition which requires the positive half-space of surface S0:
```
C1 = openmc.Cell(name="/Solid11", region=((+S450 & -S451 & +S76 & -S75) | (-S736 & +S450 & +S75 & -S77) | (+S0 & -S117 & -S116 & +S79 & -S78)
```

# Expected behaviour
I expected for the cells to not reference cell S0 - I've seen in previous conversion files by GEOUNED the surface indexing starts at 1.

# Error message
If applicable, include the error message displayed in the console: strange type of surface, unexpected exit, python
exception…
No error is thrown by GEOUNED. But upon loading the file using openmc.Geometry.from_xml, this error is thrown by OpenMC:

```
    model.geometry = openmc.Geometry.from_xml(path = geometry_path)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/raska/miniconda3/envs/openmc/lib/python3.12/site-packages/openmc/geometry.py", line 300, in from_xml
    return cls.from_xml_element(root, materials)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/raska/miniconda3/envs/openmc/lib/python3.12/site-packages/openmc/geometry.py", line 254, in from_xml_element
    c = openmc.Cell.from_xml_element(e, surfaces, mats, get_universe)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/raska/miniconda3/envs/openmc/lib/python3.12/site-packages/openmc/cell.py", line 759, in from_xml_element
    c.region = Region.from_expression(region, surfaces)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/raska/miniconda3/envs/openmc/lib/python3.12/site-packages/openmc/region.py", line 124, in from_expression
    tokens.append(+surfaces[abs(j)])
                   ~~~~~~~~^^^^^^^^
KeyError: 0
```

# Please complete the following information):
 - OS: Description:	Ubuntu 22.04.4 LTS
 - GEOUNED version: 1.2.0
 - Python version: 3.12.5

# Additional context
Add any other context about the problem here.

raska-s, 2024-08-22:
Trying with 
```
mySettings = geouned.Settings(
		startCell = 0,
		startSurf = 0)
```
Also does not generate a surface starting from 0

akolsek, 2024-08-23:
Hey @raska-s, could this be your issue: [issue 269](https://github.com/GEOUNED-org/GEOUNED/issues/269#issuecomment-2284318291), [issue 201](https://github.com/GEOUNED-org/GEOUNED/issues/201)? Usually the presence of splines causes this issue.

raska-s, 2024-08-23:
Thanks for this! Can confirm the CAD file contains splines. Is it possible to add a feature that stops conversion when splines are detected? Or a pre-check before solid decomposition?

psauvan, 2024-08-23:
> Is it possible to add a feature that stops conversion when splines are detected? Or a pre-check before solid decomposition?

yes, I am working on it.